Nikolay-Kha/PyCNC #5: PI zero w

budwilson, 2017-07-24:
It doesn't seem to work with pi zero w.  Can it be updated to run on pi zero?

Nikolay-Kha, 2017-07-25:
It can be updated. But I don't have RPi Zero or even RPi1(there is the same SoC like in zero) to test it.
Most probably there is some issue(or lack of some revisions) in detecting board version code here - https://github.com/Nikolay-Kha/PyCNC/blob/master/cnc/hal_raspberry/rpgpio_private.py#L15
Try to add '9000c1', '900092', '900093' revisions there.

Also try to run rpgpio.py file separately from the whole project: connect LED to board pin(#21 by default) and check that this test - https://github.com/Nikolay-Kha/PyCNC/blob/master/cnc/hal_raspberry/rpgpio.py#L414 works. It should blink on start using direct access to GPIO ports and then it should test DMA+GPIO. If LED doesn't blink during some mode - then it need further investigation. In case of 'dmagpio' mode fail, try to change DMA channel here - https://github.com/Nikolay-Kha/PyCNC/blob/master/cnc/hal_raspberry/rpgpio.py#L85 to 3, 5 and 6.

budwilson, 2017-07-26:
> Kha/PyCNC/blob/master/cnc/hal_raspberry/rpgpio_private.py#L15
> raspberry/rpgpio.py#L414 works. It should blink on start using direct
> Kha/PyCNC/blob/master/cnc/hal_raspberry/rpgpio.py#L85 to 3, 5 and 6.

Thanks so much for your reply.  I did not expect such a rapid response.
Below is the error message I am receiving. I tried the mods you suggested,
no effect.  I haven't tried connecting a LED yet.

I think your concept of using DMA is brilliant.

Thanks

Bud

>>>

---- ads111x is not detected ----

Traceback (most recent call last):

  File "/home/pi/PYCNC/PyCNC-master/pycnc", line 3, in <module>

    import cnc.main

  File "/home/pi/PYCNC/PyCNC-master/cnc/main.py", line 10, in <module>

    from cnc.gmachine import GMachine, GMachineException

  File "/home/pi/PYCNC/PyCNC-master/cnc/gmachine.py", line 4, in <module>

    from cnc import hal

  File "/home/pi/PYCNC/PyCNC-master/cnc/hal.py", line 101, in <module>

    from cnc.hal_raspberry.hal import *

  File "/home/pi/PYCNC/PyCNC-master/cnc/hal_raspberry/hal.py", line 10, in
<module>

    gpio = rpgpio.GPIO()

  File "/home/pi/PYCNC/PyCNC-master/cnc/hal_raspberry/rpgpio.py", line 21,
in __init__

    self._mem = PhysicalMemory(PERI_BASE + GPIO_REGISTER_BASE)

  File "/home/pi/PYCNC/PyCNC-master/cnc/hal_raspberry/rpgpio_private.py",
line 117, in __init__

    fd = self._open_dev("/dev/mem")

  File "/home/pi/PYCNC/PyCNC-master/cnc/hal_raspberry/rpgpio_private.py",
line 129, in _open_dev

    fd = os.open(name, os.O_SYNC | os.O_RDWR)

PermissionError: [Errno 13] Permission denied: '/dev/mem'

>>>

Nikolay-Kha, 2017-07-26:
This particular error says that PyCNC can open physical memory because there is no access rights.
PyCNC should be run with superuser privileges. Just run it with 'sudo'.
sudo ./pycnc

goatchurchprime, 2018-01-06:
I can't get any response from a PiZero on this.  Has there been any known progress?  

I'll look for Py 1B when I can find it in the cupboard and try this out

budwilson, 2018-01-06:
Julian,

I did get my pi zero w working.  It worked very well. I didn't hook it to
motors but I did connect diodes for the motor commands and ran all of
Nikolay's test samples.  The Zero works surprisingly well.  It is not as
fast as the PI 3.  It does have respectful speed.  I was in the mist of
designing my own CNC from scratch. All individual materials exceeded $10
each.  Thus if I build the CNC I will upgrade to the PI 3.  The PI 3 offers
a lot more flexibility for $35.

Good Luck

Bud

goatchurchprime, 2018-01-08:
I'm getting a good response of pulses on a borrowed Pi3 Model B Vi 2, but my PiZero doesn't work at all.  

Often it hangs in a loop in cnc/hal_raspberry/hal.py waiting for  dma.is_active() to return False.

I cannot see any settings for enabling/disabling dma features (as there are for i2c).  

Do you have any more details for how you got this to work on the Zero?  

Is there a more diagnostic-centric version of rpgpio.py that can better locate the point of failure?

budwilson, 2018-01-09:
sorry can't help.  Make sure the software is identical between the two
devices.

Bud