Nikolay-Kha/PyCNC #6: Pyboard support

Martronic-SA, 2017-08-06:
That projet seems ideal to work on pyboard? Any chance to see it ported to micropython?

Nikolay-Kha, 2017-08-07:
This project aims to run with ARM Cortex-A architecture processor and Linux environment.
Of course, it's possible to use a DMA module of processors with Cortex-M architecture(which pyboards mostly are) architecture. And live without Linux(though some modifications in micropython to have direct access to physical addresses might be needed).
Actually, it's even better to use microcontroller approaches to make a CNC controller with pyboard. Just disable automated garbage collector, everything else that might interrupt main thread and switch gpios in real time...
But, honestly, I don't see any reasons to do that. Just look at CPU frequencies, number of cores, RAM size of boards etc with Cortex-M and Cortex-A processors... And then take a look at prices - pyboards are hardly cheaper than raspberry pi. Abilities which provides Linux and such powerful processors are incomparable with tiny Cortex-M.

Martronic-SA, 2017-08-08:
The idea was to replace the Arduino/Marlin and other similar setups with something 100% python.

khessels, 2019-01-19:
I don't know if its the right place to ask, but i just created a duet based 10 driver controller (with spi) based on the duet schematics. I kept the mcu replaceable so i can swap it for an stm32f405 (for micropython) is it possible to elaborate here or somewhere else what would be needed to get it running on micropython ?

[image]

Ryanshuai, 2020-09-30:
> The idea was to replace the Arduino/Marlin and other similar setups with something 100% python.
> Send Gcode to the pyboard that interprets and drives the motors (with additional A4988 drivers, end stops, ..)

hi, did you try your idea?
I also think it would be fine to work on a stm32 pyboard.
There are pyboard less than 10$ in China.

khessels, 2020-09-30:
No, the project evolved, now i'm using klipper with a btt gtr

Ryanshuai, 2020-10-01:
> No, the project evolved, now i'm using klipper with a btt gtr. I am toying with the idea to enhance micropython to allow it to have g\m codes in the python source... The reason being, that i'm very much into tool-changing and adding functionality \ intelligence to my motion systems, and this would allow me to add functionality rather easy..but i'm a bit low on available time.. :( Besides, klipper\rpi is also a very interesting platform, and i might be able to do it there).

Sounds really cool, do you have a repository for it? I want to see it.

khessels, 2020-10-02:
lol, no, not right now, im building my tool changer for my new
3dprinter\cnc, after that i am going to start with that.
Sorry :)