Nikolay-Kha/PyCNC #7: Servo motor control

mircoslepko, 2017-08-24:
Hi Nikolay, really a nice project, congratulations!
I would like to make some modifications to have the ability to control a small servo motor using the command M3 S([0-9]+)

In the **hal.py** file I find spindle_control() that I could use for my purpose.
According to you, which pin can I use (and how) to rotate the servo?
I would like **M3 S0 servo motor = 0° and M3 S10000 servo motor = 180°**
I see that you manage everything through the **rpgpio.py** file, I was thinking I could add a new method (eg. add_pin_servo() ) but I do not know how to handle gpio for servo rotation.

My need is to use the servo to move up and down a pen (am i making a plotter) do you have any alternative solution to handle a servo?

Can you give me some help?
Thank you for the reply

Nikolay-Kha, 2017-08-24:
Hey, 
thank you for interest in PyCNC.
You don't need to do anything :) Just connect you servo to GPIO4 and it should work.
The trick is that you need to use M3 command which is already implemented for spindle control by PWM. It generates PWM with duty cycle form 0%( on S0) to 100%(on S10000 which is maximum rpm according config file - https://github.com/Nikolay-Kha/PyCNC/blob/master/cnc/config.py#L41). The same PWM you need for servo. So use it.

mircoslepko, 2017-08-24:
Thank you very much for the quick response.
Before I wrote to you I tried to connect the servo to GPIO4 but any S command i wrote to change the speed the servo motor always stood at the same point.
Now I try to change the servo then if I write to you.
thanks for now ;-)

Nikolay-Kha, 2017-08-24:
Just to check everything, try to connect LED(with resistor) instead of servo and see if brightness changes with different M3 commands(M3S2000, M3S5000 etc). If it works, servo should work too.
Feel free to contact me again.

mircoslepko, 2017-08-24:
right! 
Tonight I try and update you tomorrow

mircoslepko, 2017-08-25:
Hi Nikolay!
I'll update you on the tests I did last night ...
Connecting a led the PWM works properly, S0 led off S10000 led switched on at maximum power.
Now the problem is that if I connect a servo motor (i tried 2 servo motors) any command S the rotation angle remains unchanged.
I would like to ask you a question, did you ever try to connect a servo motor?
Soon and thank you again
regards

Nikolay-Kha, 2017-08-25:
Hi,
For this particular project I have never connected servos, but of course, I used them before. You issue might be related with voltage which raspberry pi provides. All GPIO pins provide 3.3 Volts. Your servo might require 5V for operating. Try to connect servo control wire to 3.3 Volts and ground  from raspberry to check if it moves.
Another issue might be PWM frequency. PyCNC PWM works at ~11.5KHz. Servo might require a little bit less. Try to change this value https://github.com/Nikolay-Kha/PyCNC/blob/master/cnc/hal_raspberry/rpgpio.py#L262 to 2560, it should provide 1.1 KHz PWM, or even to 25600 for 110Hz PWM.

mircoslepko, 2017-08-25:
Great! I will try to follow your directions.
thanks for your time

mircoslepko, 2017-08-25:
One last thing to be precise
The servo I use is SG90 9g and should work at 50Hz.
regards

Nikolay-Kha, 2017-08-25:
Try to specify 32768 here - https://github.com/Nikolay-Kha/PyCNC/blob/master/cnc/hal_raspberry/rpgpio.py#L262
32768 should give 90Hz theoretically, and theoretically servo should work with this frequency.

mircoslepko, 2017-08-29:
Hi Nikolay!
I want to update you on what I did.
I tried to follow your advice and I confirm that the servo motor responds well.
Another problem came out because when the stepper motors are running, they are interfering and the servo starts to rotate at random.

I tried to isolate the DMAPWM methods by creating 2 (1 x the stepper and 1 x the servant but nothing) but the problem remains. I've tried her all the solutions.

The only stable solution I adopted was to connect to the raspberry (via USB) an arduino and I modified your spindle_control function in hal.py that connects to the serial port of arduino and write the percentage via serial, then with a Small skecth I manage one or more servo motors and much more.
I wanted to thank you for all your kindness and kindness.

When I finish the project I will do a post on instructable and I will also refer to your excellent project (if you agree obviously).
It's a bit blurry but in the meantime I leave you this picture

[image]

Thanks again
regards
Mirco

Nikolay-Kha, 2017-08-29:
Hello Mirco,
I'm happy that you manage to run it. Of course, feel free to refer, share links or what ever you want about PyCNC project. And I'll will be happy if you share a link to your future post here.

Meanwhile, I think I know the reason why servo behaves unstable with steppers, I'm not sure how to fix it and when I will do this, but I will think about it and will add servos support in future releases. So I will keep this issue opened until fix.

I wish you all the best with your project. And thank you for image.