Nikolay-Kha/PyCNC #9: How to run gcode command on pycnc startup

mircoslepko, 2017-09-06:
Hi Nikolay,
my project is also going to great thanks to your PyCNC !!! ;-)
Now I would like to ask you what is the best practice to run gcode commands whenever you launch command sudo ./pycnc.
I tried to add them to main.py (https://github.com/Nikolay-Kha/PyCNC/blob/cf0b4d0a72e7e42eb307ec6312ab30668ea81cec/cnc/main.py#L59)
but I do not know if it's a good thing.

There is a point in the configuration where to place commands to run at the statup? example G28.

Thanks again
regards
Mirco

Nikolay-Kha, 2017-09-07:
Hello Mirco,

Typically all CAM software has setting something like "gcode prefix" where you can put some gcode commands to initialize machine. That is a most correct way to do it.
If you haven't such setting or have just gcode files then you would need to change this two lines
https://github.com/Nikolay-Kha/PyCNC/blob/master/cnc/gmachine.py#L37
to
hal.init()
self.reset()
i.e. initialize hal at first and then call reset(). Then add at the end of reset() function actions that you would like to do, but without gcode. For example for g28 command, add 
self.safe_zero()
after one or bunch of move_xxx() commands hal.join() should be called to make sure that movement finished.

But if you need really fast solution, just add lines like
machine.do_command(GCode.parse_line("G28"))
right in the begging of main function - https://github.com/Nikolay-Kha/PyCNC/blob/master/cnc/main.py#L44

mircoslepko, 2017-09-07:
very good, thank you for the advice!