Nikolay-Kha/PyCNC #12: Can't put the PyCNC working...

eirasys, 2017-10-15:
Nikolay, hello
I've been testing the PyCNC whithout success.
Have Raspberry Pi 2 Model B Rev. 1.1, it should work on this model, right?
I've tested the file PyCNC-master/cnc/hal_raspberry/rpgpio.py as testing purpose,
just like you mentioned, after changing the first line from:
from .rpgpio_private import *
to:
from rpgpio_private import *
I can test the LED on GPIO 21, and it works
But when trying it with G commands running the application, there isn't any signal at GPIO21, the LED is always off.
DMAGPIO.run_stream is different from DMAGPIO.run!
Your code calls the first and never the second, while the LED test calls the run function.
Give me some hints to find where is the problem,
Thanks,
João

eirasys, 2017-10-15:
Just tried to change the DMA channel to 3, 5 and 6 instead of the default 4. Just like your answer to Pi zero issue. No success..
Any help will be great, thanks

eirasys, 2017-10-16:
Problem solved!!
I'm using DM860A, the problem is the STEPPER_PULSE_LENGTH_US,
change it from 2us to 55us, less than this value and it will not respond properly.
PyCNC works very well with DMA having very smooth pulses.
Problem is time spend in generator and sometimes the wait while DMA active.
Maybe RP3 can solve that..

Nikolay-Kha, 2017-10-20:
Hello.
Nice to know that problem is solved. If you experience problem in long delays before motors actually start moving on each command, try to use pypy( https://github.com/Nikolay-Kha/PyCNC#performance-notice )

eirasys, 2017-10-25:
Hello,
Added this line to the main:
# Interpreter Version
print('Python Interpreter Version:', sys.version)
I can see some difference with pypy, the motors start moving faster but sometimes there is a big delay in the end of the move. Can't understand why since all processing was already done.
Thanks