Nikolay-Kha/PyCNC #13: How to sync motion with I2C PWM for Laser Cutter/Engraver

eirasys, 2017-10-26:
Hello Nikolay
Have a PCA9685 16ch 12bit PWM controlled by I2C bus. I'm using it to control
a laser cutter/engraver with higher PWM resolution. Is it possible to sync DMA motion
with I2C messages, and how fast can it go?
I see that you are using I2C to measure temperature with an ADC, but this speed will surely
be quite slower than what I need.
Thanks,
João

eirasys, 2017-11-01:
Planing to add a new axis to hal.move(generator).
This axis may have pulse information for the laser to cut or engrave anything.
I2C may result in inacurate sync with motors motion.

Nikolay-Kha, 2017-11-03:
Hello,
Theoretically DMA can write data to I2C bus, and since DMA switching take the same number of transferred bytes with the same frequency, it should be able to sync with motors.
Though it is not quite simple to implement: you need to increase laser power smoothly while motors accelerates...
You second solution about additional axis is much better. Just create a proper PWM generator for this special axis in pulses.py instead of motor pulses. It should work fine.

Nikolay-Kha, 2017-11-03:
and one more thing - you can achieve even better resolution then 12 bit using this technique, the only one thing you will be limited with PWM frequency. The bigger resolution, the lower PWM frequency will be.

eirasys, 2017-11-30:
Fully agree! There is a lot of work to be done, different PWMs defining laser power levels and photo rastering conversion to PWM for engraving purposes.

openbuilds-engineer, 2017-11-30:
For the conversion see
github.com/LaserWeb/LaserWeb4-Binaries/releases/latest (;

My old project...