Nikolay-Kha/PyCNC #15: lcd support

ekaggrat, 2017-11-22:
will you be adding lcd support any time soon?

Nikolay-Kha, 2017-11-26:
That is a very old fashion... Raspberry Pi has HDMI output and OS with fully functional graphic interface. Even with OpenGL support.
So, exactly LCD support will never be added. 
Probably HTTP web server will be embedded into this project. It will allow to display web interface on remote machines or right on the same rpi3 using browser.

ekaggrat, 2017-11-26:
What I meant was lcd interface on the rpi lcd itself. Good that it will be added .

eirasys, 2017-11-30:
Yes, web interface is the way to go, I'm trying to develop it:
 https://github.com/eirasys/PyCNC-with-WEB
It already works but it is a little bit slow when compared to pypy execution.
Performance is a problem because WebIOPi is using a different Python interpreter that I don't know if its possible to change to Pypy.