Nikolay-Kha/PyCNC #16: Problem with external heatingsoftware

jamadir, 2017-11-30:
Hello
I build a custom 3d Printer from scratch, and want to controll its Temperatures with an external programm.
I use an mcp3008 to convert the analog Thermistor-Signal to digital and read it out using an external programm. The Problem is, that if i s

tart pycnc the temperature-values just jump around and aren't usable. 
I am fairly new to programming, especially in python and don't know how to solve the Problem myself. 
I used print and time.sleep commands to track the command causing the Trouble down. In the main.py document  in line 27 the at the Command machine = GMachine() the readings go nuts.
i really hope you can help me, its very frutrating to nearly complete a projekt and get stuck at this Problem.

(sorry for my grammar)
[image]

Nikolay-Kha, 2017-12-01:
Hello,
To measure temperature by external software you would need to re implement this method - https://github.com/Nikolay-Kha/PyCNC/blob/master/cnc/sensors/thermistor.py#L53
Just run whatever you want in this method and return float value in Celsius.

jamadir, 2017-12-01:
first of all, thank you for your time, i did what you said and implementet my code into the thermistor.py file 
[image]
but even now, if i read it out while the software is running, the readings make no sense:
[image]
the voltage should be 3.16 volts, and here its jumping between 0.4 and 1.4
this is the code which gives me the right voltages 
[image]
[image]
I hope you can identify the problem.

Nikolay-Kha, 2017-12-02:
The only one difference I can see. There is no initialization of SPI pins(GPIO.setup()). Also please make sure that you've selected proper pins and they don't conflict with other pins from config file.

If initialization doesn't help, please add:
print(adc)
in both projects right after spi.xfer2(). Let's compare if data is the same.

jamadir, 2017-12-03:
Thank you, i had my fan pin assinged to one of the ADC pins, i would never have thought of that. Now it gives out the right temperatures an i finally can start printing :D your sofware is great, thank you for your time

with greetings Jannes

closed 2017-12-03 by jamadir