Nikolay-Kha/PyCNC #20: Error occur when running with pypy

Wannawuth, 2018-03-07:
Hello Nikolay,

Thank you for your sharing 

My system is consisting of:
Hardware
   Raspberry Pi3 Model B
Software
   Linux raspberrypi 4.9.59-v7+ #1047
   Python 2.7.13
   Pypy2-v5.7.1-linux-armhf-raspbian

PyCNC can running with python. I test with command G1 and I get a low speed.

When pycnc running with pypy, an error occur as follow:
[image]

May I get your advice?

yo-k, 2018-03-10:
well, seems to be a pypy bug up to <5.7 (and maybe again on 5.7.1???). I have tried to install 5.10 https://bitbucket.org/pypy/pypy/downloads/pypy2-v5.10.0-linux-armhf-raspbian.tar.bz2 but got new errors  about missing libs. As I updated the kernel via rpi-update to 4.14x and I guess some libs that may be the problem.

@Wannawuth could you try to install that pypy version instead of 5.7.1 on your machine and post the result?

```
sudo rm /usr/local/bin/pypy #removing old link from 5.7.1
sudo rm /opt/pypy/* #removing old pypy version in that directory
wget https://bitbucket.org/pypy/pypy/downloads/pypy2-v5.10.0-linux-armhf-raspbian.tar.bz2
sudo tar xvf pypy2-v5.10.0-linux-armhf-raspbian.tar.bz2 --directory /opt/pypy/ --strip-components=1
sudo ln -s /opt/pypy/bin/pypy /usr/local/bin/pypy

```
PS: I have tested pycnc on rpi3 instead of rpi1, now. Much better!!!! :-)

Nikolay-Kha, 2018-03-12:
Hello,
I used PyPy v5.7.1 and everything worked for me :) So try to install this version:
```bash
wget wget https://bitbucket.org/pypy/pypy/downloads/pypy2-v5.7.1-linux-armhf-raspbian.tar.bz2
sudo mkdir /opt/pypy
sudo tar xvf pypy2-v5.7.1-linux-armhf-raspbian.tar.bz2 --directory /opt/pypy/ --strip-components=1
sudo ln -s /opt/pypy/bin/pypy /usr/local/bin/pypy
```

Wannawuth, 2018-03-12:
I was installed pypy2-v5.10.0
When pypy running, I get an error message: error while loading shared libraries: libssl.so.1.0.0 cannot open shared object file: No such file or directory 
Then I was downloads and install libssl1.0.0_1.0.1t-1+deb8u7_armhf.deb

Now I get an error message: error while loading shared libraries: libffi.so.5: cannot open shared object file: No such file or directory 

I think pypy v5.0+ can run pycnc. 
Now problem is how to install pypy?
With my limited of the knowledge in linux, It may be please if someone can help!

Nikolay-Kha, 2018-03-12:
@Wannawuth try `sudo apt install libffi5` or `sudo apt install libffi6`.

yo-k, 2018-03-12:
@Nikolay-Kha :  Which version of raspbian/debian do you use? I am using stretch, not jessie. It has more up to date versions of libssl, libffi etc.  May be that is the problem?

Wannawuth, 2018-03-13:
When use  `sudo apt install libffi5 `
result:  `E: Unable to locate package libffi5`

When use  `sudo apt install libffi6 `
result: `libffi6 is already the newest version (3.2.1-6).`
           `0 upgraded, 0 newly installed, 0 to remove and 153 not upgraded. `

Then run `sudo pypy --version`
result: `pypy: error while loading shared libraries: libffi.so.5: cannot open shared object file: No such file or directory`

Wannawuth, 2018-03-13:
After I manual install _libffi5_3.0.10-3+deb7u2_armhf.deb_
Now pypy can running and pycnc get better motor speed.

Thank you for all advise

Nikolay-Kha, 2018-03-14:
@yo-k during development it was raspbian from the official web site https://www.raspberrypi.org/downloads/raspbian/
Image was downloaded one year ago, I don't remember exact version, but it was definitely jessie.

yo-k, 2018-03-26:
@Nikolay-Kha Thanks for answering. I am still trying to get pypy working with up to date libs but did not have enough time to dig deeper. I think it is a real pypy issue - not a pycnc issue. But I don't know.

Wannawuth, 2018-03-27:
@yo-k When pypy running, I get an error message: _error while loading shared libraries: libssl.so.1.0.0 cannot open shared object file: No such file or directory_ 

I was downloads a file _libssl1.0.0_1.0.1t-1+deb8u7_armhf.deb_ from a link: [http://ftp.nl.debian.org/debian/pool/main/o/openssl/libssl1.0.0_1.0.1t-1+deb8u7_armhf.deb)](url)
Then I get a file: _libssl1.0.0_1.0.1t-1+deb8u7_armhf.deb_
Then manually install by running: _sudo dpkg -i libssl1.0.0_1.0.1t-1+deb8u7_armhf.deb_

Next run pypy, I get new error message: _error while loading shared libraries: libffi.so.5: cannot open shared object file: No such file or directory_

I was downloads a file _libffi5_3.0.10-3+deb7u2_armhf.deb_ from a link:
[http://ftp.nl.debian.org/debian-security/pool/updates/main/libf/libffi/libffi5_3.0.10-3+deb7u2_armhf.deb](url)
Then I get a file: _libffi5_3.0.10-3+deb7u2_armhf.deb_
Then manually install by running: _sudo dpkg –i libffi5_3.0.10-3+deb7u2_armhf.deb_ 

Finally pypy is work!

yo-k, 2018-03-27:
@Wannawuth Thank you! It is a work around but depends on old security relevant libs.
@Nikolay-Kha Why do pycnc needs libssl? Or why is pypy checking for libssl?

Nikolay-Kha, 2018-04-02:
@yo-k PyCNC don't neew libssl. I think pypy needs it because it is probably dynamically linked with pypy binaries.