Nikolay-Kha/PyCNC #21: E Motor

hmovahedim, 2018-03-10:
hello Nikolay
how can i use extruder and this motor in my G_Code ?

Nikolay-Kha, 2018-03-12:
Hello.
Yes, you can. You can heat up extruder with M104 or M109 command. M104 command exits immediately, M109 waits until extruder reach specified temperature.  Temperature can be specified in S argument. It should be in Celsius degrees.  Example:
m109s200
this command will heat up extruder up to 200 C. During this time other commands wouldn't be allowed.
To move extruder, you need to use G1 movement like for other axises. Position of extruder motor is absolute, so you need to increase value with each command.
Example:
g1e1
this command would melt 1mm of filament. To move on the next 1mm you need to do this command:
g1e2
That is basically it. Hope this helps. But I don't understand - why do you need to write GCode manually? There are plenty of cool CAM programs which would prepare GCode for you automatically. Slic3r ( http://slic3r.org/ ) for example. There is a config for this program and PyCNC notation of gcode - https://github.com/Nikolay-Kha/PyCNC/blob/master/extra/sample-Slic3r-config.ini

hmovahedim, 2018-03-12:
thanks sir.
this is helpful.
i want to learn more about pycnc.

hmovahedim, 2018-03-13:
my extruder isn't working good because my extruder motor is working bad and grind my fillament . 
how can i config this ?

Nikolay-Kha, 2018-03-14:
What do you mean by "motor is working bad" and "grind filament"? Do you have issue with pulse generation for motor? Does motor work find on idle(without filament)? Have you properly calculated amount of filament to use with axises motors?

hmovahedim, 2018-03-15:
thanks sir.