Nikolay-Kha/PyCNC #22: How PyCNC work with a long travel length machine?

Wannawuth, 2018-03-26:
Hello,

In config.py I change:

#Stepper motors step per millimeter for each axis.
STEPPER_PULSES_PER_MM_X = 400
STEPPER_PULSES_PER_MM_Y = 400
STEPPER_PULSES_PER_MM_Z = 400

#Workplace physical size.
TABLE_SIZE_X_MM = 2500
TABLE_SIZE_Y_MM = 1300
TABLE_SIZE_Z_MM =   250

When I running PyCNC for a travel length <150mm.
The motor are moving well.

But, when I running in a travel length > 250mm.
The motor are moving until I get _"MemoryError: Out of allocated memory."_

There are any possible ways to overcome this? 
How to specify a maximum travel length?

yo-k, 2018-03-26:
My test machine setup got 100 steps/mm. The edge is at 2457.60mm or 245760 steps.  2457.61mm do not work.

```
*************** Welcome to PyCNC! ***************
> G0 X0
OK
> G0 X2457.60
OK
> G0 X0
OK
> G0 X2457.61
Traceback (most recent call last):
  File "./pycnc", line 6, in <module>
    cnc.main.main()
  File "/home/pi/pycnc/PyCNC/cnc/main.py", line 64, in main
    do_line(line)
  File "/home/pi/pycnc/PyCNC/cnc/main.py", line 33, in do_line
    res = machine.do_command(g)
  File "/home/pi/pycnc/PyCNC/cnc/gmachine.py", line 361, in do_command
    self._move_linear(delta, vl)
  File "/home/pi/pycnc/PyCNC/cnc/gmachine.py", line 126, in _move_linear
    hal.move(gen)
  File "/home/pi/pycnc/PyCNC/cnc/hal_raspberry/hal.py", line 280, in move
    dma.add_delay(k - prev)
  File "/home/pi/pycnc/PyCNC/cnc/hal_raspberry/rpgpio.py", line 160, in add_delay
    raise MemoryError("Out of allocated memory.")
MemoryError: Out of allocated memory.
```

At 200 steps / mm 245760 steps would be 1228.80mm,  1228.805mm should not work. Voilà:

```
*************** Welcome to PyCNC! ***************
> G0 X0
OK
> G0 X1228.8
OK
> G0 X0
OK
> G0 X1228.805
Traceback (most recent call last):
  File "./pycnc", line 6, in <module>
    cnc.main.main()
  File "/home/pi/pycnc/PyCNC/cnc/main.py", line 64, in main
    do_line(line)
  File "/home/pi/pycnc/PyCNC/cnc/main.py", line 33, in do_line
    res = machine.do_command(g)
  File "/home/pi/pycnc/PyCNC/cnc/gmachine.py", line 361, in do_command
    self._move_linear(delta, vl)
  File "/home/pi/pycnc/PyCNC/cnc/gmachine.py", line 126, in _move_linear
    hal.move(gen)
  File "/home/pi/pycnc/PyCNC/cnc/hal_raspberry/hal.py", line 280, in move
    dma.add_delay(k - prev)
  File "/home/pi/pycnc/PyCNC/cnc/hal_raspberry/rpgpio.py", line 160, in add_delay
    raise MemoryError("Out of allocated memory.")
MemoryError: Out of allocated memory.

```

Back again to 100 steps / mm. Trying the steps with X and Y.

```
*************** Welcome to PyCNC! ***************
> G0 X2457.60 Y0
OK
> G0 X0 Y0
> G0 X2457.60 Y1
Traceback (most recent call last):
  File "./pycnc", line 6, in <module>
    cnc.main.main()
  File "/home/pi/pycnc/PyCNC/cnc/main.py", line 64, in main
    do_line(line)
  File "/home/pi/pycnc/PyCNC/cnc/main.py", line 33, in do_line
    res = machine.do_command(g)
  File "/home/pi/pycnc/PyCNC/cnc/gmachine.py", line 361, in do_command
    self._move_linear(delta, vl)
  File "/home/pi/pycnc/PyCNC/cnc/gmachine.py", line 126, in _move_linear
    hal.move(gen)
  File "/home/pi/pycnc/PyCNC/cnc/hal_raspberry/hal.py", line 281, in move
    dma.add_pulse(pins, STEPPER_PULSE_LENGTH_US)
  File "/home/pi/pycnc/PyCNC/cnc/hal_raspberry/rpgpio.py", line 131, in add_pulse
    raise MemoryError("Out of allocated memory.")
MemoryError: Out of allocated memory.
```

Seems to be a limit at 245760 steps at one time. But it isn't. Here are 245780 steps:

```
*************** Welcome to PyCNC! ***************
> G0 X1228.8 Y1229
OK
```

I don't know... error in reasoning?

Nikolay-Kha, 2018-04-02:
Hello. Yes, you are right - there is a limit for number of steps. That's why big table doesn't work with precise steppers(400 pulses per mm). To increase buffer size, you can try to change amount of allocated memory here - https://github.com/Nikolay-Kha/PyCNC/blob/master/cnc/hal_raspberry/rpgpio.py#L97
Memory is allocated from GPU memory and by default there is a limit - 64 MiB. If you don't use monitor, you can try to take whole this memory. If you need monitor or more memory - you need to adjust GPU memory size in system, see how to do that here for example - https://www.raspberrypi.org/forums/viewtopic.php?t=190182

Wannawuth, 2018-04-05:
Thank you! Your advices are most important. 
After adjust the precise steppers and increase buffer size, now PyCNC can run with a large size works. 

I think PyCNC is high flexibility use in a wide works.