Nikolay-Kha/PyCNC #29: Regarding use of Rockpro64 with pycnc

bharathbellary, 2018-10-29:
Hello nikolay,
        What a wonderful project you have done it's nice and useful .
But i'm facing a problem with using the pycnc with my new Rockpro64 i tried different methods to port it but it was unsuccessful please guide me to how to use pycnc with RockPro64
           Thank you

Nikolay-Kha, 2018-11-28:
Hello
I'm happy to hear that project is useful.
Current implementation exists only for bcm283x processors which are used in Raspberry Pi boards. RockPro64 board uses Rockchip RK3399 processor. And to use it with PyCNC, hardware abstraction layer(HAL) should be implemented for PyCNC. Here is the HAL for RPI - https://github.com/Nikolay-Kha/PyCNC/tree/master/cnc/hal_raspberry

bharathbellary, 2018-11-28:
> Kha/PyCNC/tree/master/cnc/hal_raspberry

Hello,
Thank you very much for your reply your project is awesome and I'm trying
it out with my different systems as I'm not weak in programming I'm not
able to using your project with Rock pro can you please guide me how to get
this working with rockpro 64?

Nikolay-Kha, 2018-11-28:
You need to implement the same thing which is implemented for RPi here - https://github.com/Nikolay-Kha/PyCNC/tree/master/cnc/hal_raspberry
Prototype for this class - https://github.com/Nikolay-Kha/PyCNC/blob/master/cnc/hal.py
Another, virtual, i.e. dummy implementation - https://github.com/Nikolay-Kha/PyCNC/blob/master/cnc/hal_virtual.py

You would need to find a way to use Rockchip DMA controller to generate pulses, control GPIO ports etc. As far as I remember that processor has even better then RPi DMA module and create this should be even easier, but still require a lot of thinking and reading of chip documentation.

Nikolay-Kha, 2018-11-29:
Just to change addresses is not enough. DMA modules are different. You need to find a way to start DMA module to copy data from buffer to GPIO module. Registers and possible values are described in chip documentation. To explain that to DMA module you would need to write something in DMA registers.
You could start with simple GPIO. Try to write data directly to GPIO module to control pins. Try to blink with LED. See this lines: https://github.com/Nikolay-Kha/PyCNC/blob/master/cnc/hal_raspberry/rpgpio.py#L415-L430 - that is a test for RPi to blink with LED on pin 21. It also needs to be implemented for HAL. And that is a good start to understand how things work in processor registers.

bharathbellary, 2018-12-01:
Thank you very much for your response yup I'll try these and let you know
and thank you for the project it's was awesome 😊✌️