Nikolay-Kha/PyCNC #31: Complete explanation about defining 4th axis as laser engraving please

pnumaple, 2018-11-23:
Hi dear nikolay. i'm hossein...

i'm so excited from your project. i'm your big fan ... recently i test my GRBL LASER WITH Arduino and the speed of laser engraving for vector files was good , but for rastor files was dramaticly bad , because low buffer and serial connection for arduino Uno. when i saw your project , it makes so excited and the mechanism for for solving real time is fantastic. i have some question...
1) i understand the main concepts about your explanation and thank you so much to your big project...thats awesome... but as you mentioned in comments, defining 4th axis is the best way to use laser ...
please explain more preciely and step by step for changes in files and pins that have been used...

2) i follow step by step your flowchart to use laser instead if spindle and write G1 S100 when i connect the TTL wire of my laser to GPIO 4 , but nothing was happend... which pi is associted for this? thank you so much

Nikolay-Kha, 2018-11-28:
Hello Hossein,
You would need to add additional axis to your gcode. For example 'L'.
So commands with laser would looks like:
G1X10L100
which would mean that movement should be performed with 100mW(may be even W) power on laser.
You don't have to modify anything in gcode parsing sources, you need to read this axis for any movements in gmachine.py like:
laser_power = gcode.get('C', self._laser_power)
and then pass this argument to interpolation functions. And here is the hardest part. You would need to simulate PWM for your laser according this power and mathematical functions. Then PyCNC convert this to real time pulses with respect to motors acceleration/breaking to laser power. To do this - try to understand how Linear interpolation works here - https://github.com/Nikolay-Kha/PyCNC/blob/master/cnc/pulses.py#L250
Hope this helps.

pnumaple, 2018-11-30:
Thank you so much dear nikolay.
Yesssss.... that's very helpful. very precise and good expalin.... 
Thanks my bro...