Nikolay-Kha/PyCNC #37: Использование концевого выключателя.

ostap-dm, 2019-05-17:
Доброго времени суток! Разбирался с вашим интерпретатором g-code, но сталкнулся с проблемой работы ENDSTOP_PIN_X. Имею обычные концевые выключатели и ни как не могу поять как их к RPi3 подключить. Получается что pin_x(23BCM) должен подтягиваться к 3,3 вольтовому разъёму RPi когда ось в положении 0? А если на пин не приходит никакой сигнал, то интерпретатор начинает уводить ось до тех пор, пока не появится 3,3 вольта? Спасибо за ответ.

Nikolay-Kha, 2019-05-29:
Доброго! По дефолтным настройкам проекта концевики должны соединять землю с пином и ставится в нулевом положении, т.е. когда станок заходит в ноль на пине должна быть земля. +3.3 подводить не надо - там подтяжка включена.

ostap-dm, 2019-05-29:
Спасибо! С этим получилось разобраться. Просто нужно было как я понял добавить функцию calibrate в код. 
Появился ещё один вопрос. На сколько сложно будет реализовать работу концевиков в качестве emergency stop. То есть при срабатывание концевика двигатели уходят в начальное положение. Понимаю логику, но достаточно сложно понять как это реализовать в коде. Возможно у вас будет возможность помочь. И сколько это будет стоить если возможность есть?

ostap-dm, 2019-05-29:
И так же вопрос по добавлению 4 и 5 осей. На сколько глобальны изменения. И какие файлы необходимо изменить для них?
Спасибо

Nikolay-Kha, 2019-06-04:
Оси добавлять в gmachine, там по идее не сложно, форки репозитория вроде были, народ добавлял оси там.
А вот Е-СТОП с возвратом организовать будет трудно, т.к. сделать остановку по срабатыванию концевика не так уж и трудно, но вот понять где он остановился будем проблемно - нужно проверять сколько DMA импульсов успел пустить - по значению регистра указателя, нужно посчитать в буфере сколько было импульсов и тогда уже двигать в начальное положение.