Nikolay-Kha/PyCNC #40: Fresadora con Pycnc no puede manejar coordenadas negativas

Gerardo1968, 2019-05-28:


Gerardo1968, 2019-05-28:
He construido una fresadora cnc basada en Pycnc y me encuentro con dos problemas:
1 El primero es que Pycnc no admite coordenadas negativas tales como G1 X-20. por lo que todo el código debe tener coordenadas positivas G1 X20.
2º He diseñado un entorno gráfico para Pycnc en el que lanzo pycnc y pierdo el control, una vez lanzado no tengo forma de pararlo a no ser parando el proceso o apagando la impresora.

Gerardo1968, 2019-05-28:
He construido...

Nikolay-Kha, 2019-05-29:
¡Hola!
Yo soy estudiante de español durante 2 meses. Es mi primera vez para usar Español publicamente :).
1. Veridico. Nunca se creo en usar las negativas. Yo creo que es facil de implementar.
2. Ejecutar la función de python machine.release () para parar completamente. No hay pausa porque esto es mucho dificil.

Gerardo1968, 2019-06-03:
Muchas gracias Nikolay.