Nikolay-Kha/PyCNC #41: How to test if my setup works?

janoist1, 2019-08-08:
Hi,
I'm very excited about this project! I have rebuilt my chinese Anet a8 printer according to the article. When I feed a gcode file to `pycnc` the printer starts to do things: making some beep-beep noise as well as heating the bed then the extruder head. And this is pretty much the progress it has done so far. I've got some questions:
- how to test each axis in `pycnc` shell?
- how to set "home" all 3 axis? I used to have an option in Merlin's menu something like "home" which aligned things to start position.

Many thanks!

Nikolay-Kha, 2019-09-06:
run pycnc in the terminal and use regular gcode.
g1x100 for example to move x only axis.
g28 to move all axis to home position.

janoist1, 2019-09-11:
Hi! Thanks for the answer. I have managed to move the Y axis, however none of the rest seem to work. I have double checked all the wires as well as replaced the Ramps with another one due to a previous burnout of the servo motors.
When I simply swap the X - Y axis wires I can see the motors working.
Have you got any idea what's going wrong? Or please give me some advices on how to debug it. I've not been able to use my printer for ages now.
Many thanks!

closed 2019-09-18 by janoist1