Nikolay-Kha/PyCNC #43: Z motors don't work

janoist1, 2019-09-18:
Hi,
I'm testing my setup using raw g code commands, like z10.
Z motors don't seem to work, they only make some whistle-like sound when supposed to be moving. I have tried with different servos and motors too. Different servo driver does the same issue, wiring to different motors also same. Just coo.
Any idea?
I can't wait to this working on my desk...

janoist1, 2019-09-19:
Ok, I have managed to make the Z work! Had to adjust the pulse `STEPPER_PULSES_PER_MM_Y` down to 50. I'm not sure what it exactly does or how it affects the end print quality. 🤷‍♂️

janoist1, 2019-09-19:
Now the question is, how to calculate the config values for my printer? Unfortunately the motors spin too fast, especially the extruder motor. Thanks!

closed 2019-09-19 by janoist1

Nikolay-Kha, 2019-09-25:
> how to calculate the config values for my printer?
Find specs for your hardware. Or calculate these values according to your construction. Or do some empirical tests - commands have values in millimeters, compare it with actual physical distance of axis and adjust parameters.