Nikolay-Kha/PyCNC #44: "M109 S200" never finishes

janoist1, 2019-09-18:
The command `M109 S200` never finishes. However the hotend is hot. When I unplug the thermistor it crashes, next time won't even start until complete restart, saying ads1115 not detected. What's wrong? Thanks!

Nikolay-Kha, 2019-09-25:
Just curious - what was the issue?

janoist1, 2019-09-27:
No solution yet. Extruder heats up slowly but doesn't seem to be getting the aimed temperature. It is hot but not hot enough to melt the PLA.

janoist1, 2019-09-27:
After turning on debugging I can see the temperature stops around 180C at max 100% power. It slowly goes up, in 20 minutes it reaches 183C.

Nikolay-Kha, 2019-10-01:
I've seen extruders which work 12V and 24V. Are you sure that you power up your extruder with correct voltage? At 100% power it should burn in few dozens of seconds....

Nikolay-Kha, 2019-10-01:
to check it with 100% confidence - connect heater directly to PSU - will it reach more than 180C? Be careful, good extruder shall reach 400-500C in 10-20 seconds.