Nikolay-Kha/PyCNC #45: How to calculate motor setting values for my config file?

janoist1, 2019-09-19:
Still very excited.
I think I'm nat far from actually printing something. All the motors work, however they are not precise. When I play with the numbers in the config such like `STEPPER_PULSES_PER_MM_X` the distance the motors take varies.

The big question is, how to get/calculate the correct values for my setup? This is an Anet A8 chinese printer, kind of stock. Thank you for your help!

janoist1, 2019-09-23:
@Nikolay-Kha Please can you help me in this?

Nikolay-Kha, 2019-09-25:
is it your printer?
https://all3dp.com/1/anet-a8-3d-printer-review-diy-kit/  (first link from google)

if so, then(even if not - i hope you can repeat this recalculation for your hardware)
MAX_VELOCITY_MM_PER_MIN is actually "Print speed: up to 100mm/s" from specs. Convert mm/s to mm per minute. How many seconds in minute? 60. Then MAX_VELOCITY_MM_PER_MIN is 6000. In your case - for each motor.
MIN_VELOCITY_MM_PER_MIN - generally - don't touch.
CALIBRATION_VELOCITY_MM_PER_MIN - generally - don't touch. Maybe set it a little bit less if calibration doesn't work properly.
STEPPER_PULSES_PER_MM_X  is actually "XY-axis positioning accuracy: 0.012mm Z-axis positioning accuracy: 0.004mm" from specs. It is one step or pulse to motor driver. That means per one millimeter z axis needs ~80 steps per mm. Set it to 80, run command g1z100 and check that extruder actually moved on 100mm with rules. For x and y axis it is 250 steps(pulses) per mm. Do the same with ruler. This calculation might be incorrect if you use motor drivers with different microsteps config.
STEPPER_INVERTED  - if motor moves in incorrect direction, just change True on False or vice versa.
ENDSTOP_INVERTED - if motors do not stop at endstops on calibration procedure - change True on False or vice versa.
TABLE_SIZE_X_MM = 220, TABLE_SIZE_Y_MM = 220, TABLE_SIZE_Z_MM = 240 from printer specs.
STEPPER_PULSE_LENGTH_US - set it to 2.
STEPPER_MAX_ACCELERATION_MM_PER_S2 - no information in specs. 3000 might be too much for that model. Most probably it must be something between 1000-2000 for you hardware. Adjust it with experiments. Motors should not make noice while they start moving.

Anything else?

janoist1, 2019-09-25:
Many thanks for the detailed explanation! 🙏🙏🙏🙏
printer: that's correct, I've got that one
motor drivers: A4988, I'm not sure where to config microsteps. According to google there are 3 pins for this, which makes me think this is also a software level configuration.
I'm going to try things with the new values first, fingers crossed!

janoist1, 2019-09-25:
With the values below, the motors make high pitched sounds and not moving much apart from small but fast jumps.
```
MAX_VELOCITY_MM_PER_MIN_* = 6000
STEPPER_PULSES_PER_MM_X = 240
STEPPER_PULSES_PER_MM_Y = 240
STEPPER_PULSES_PER_MM_Z = 80
STEPPER_PULSES_PER_MM_E = 150
STEPPER_MAX_ACCELERATION_MM_PER_S2 = 1000
```
After setting `STEPPER_MAX_ACCELERATION_MM_PER_S2` to 100, things started to look better, the motors actually move, however not precise. `x5` is ~7cm.
Reducing it to smaller numbers didn't really make things better.

Does it have anything to with the motor driver config?

Nikolay-Kha, 2019-09-25:
You can debug everything with STEPPER_MAX_ACCELERATION_MM_PER_S2 = 100. With this value printer must move axises, not that fast as it could, but must move. Now you need to get proper movements. if x5 give you 7 centimeters movement then it is clearly wrong with number of pulses. 70mm instead of 5mm. That is too much. Check microstep config.
 A4988 has microstep configuration based on pins connected to ground or Vdd. I suggest to use 1/16 configuration - all three pins(ms1-ms3) must be connected to high level. By default they have pull down resistors on board. If you use RAMPS board - add short circuit jumpers on all connector under the drivers.

Also there is a current configuration with a variable resistor on that driver. Usually it always works for me when resistor is set to the middle. At least make sure, that resistors are not somewhere in max or min position. Though you might need to find a proper way to configure it in the Internet.

janoist1, 2019-09-27:
Adjusting the driver seemed to be a good idea, I've set all to 1/16. `x10` now ~2cm. I played with the values and ended up with the following:
```
STEPPER_PULSES_PER_MM_X = 90
STEPPER_PULSES_PER_MM_Y = 90
STEPPER_PULSES_PER_MM_Z = 30
STEPPER_PULSES_PER_MM_E = 15
STEPPER_MAX_ACCELERATION_MM_PER_S2 = 3000 # 100 was too slow
```
I'm about to try printing to see if it is at least close enough, but now facing with the [extruder heating issue](https://github.com/Nikolay-Kha/PyCNC/issues/44).

Nikolay-Kha, 2019-10-01:
Z and E number of pulses doesn't look correct. Usually z axis is more precise than X and Y, i.e. require more pulses to move for 1 mm. Double check you values with 'z10' command and ruler. The same for E axis, i.e. extruder. It should be more precise. Unscrew nozzle, do not turn on heater, wire shall move without nozzle back and forth with the motor. Check with ruler that 'e100' actually move wire for 100mm.