Nikolay-Kha/PyCNC #50: Additional degree of freedom (tilting bed)

Squanchy42, 2019-11-21:
Dear Nikolay,

thanks for sharing this awesome and astonishing project. As indicated in the title of my request, I would be glad, if you could share your approach/ideas on how to add an additional axis in order to control e.g. a tilting bed or a second extruder. I can imagine this would considerably increase its applicability.

Thanks in advance and best regards

Max

Nikolay-Kha, 2019-11-23:
Hello Max,

That's kinda typical task and some people already did it. Just have a look at forks of this project - https://github.com/Nikolay-Kha/PyCNC/network
I think you can easily find their implementation of extra axis.

Thanks,
  Nikolay

Squanchy42, 2019-11-26:
Hey Nikolay,

thanks for your answer, I will have a look at that. If I may, I would like to ask a follow-up question:
What do you think is the best way to expand the hardware configuration to support the additional stepper motors?

In the picture "PyCNC reference connection" you connect the ADS1115 board with the I2C connectors. Do you use the corresponding pins on the RPi as I2C connectors? Or did you repurpose GPIO pin 2 and 3? If they indeed function as I2C connectors, I was thinking of an I2C-capable I/O expander (e.g. MCP23017) or I2C-capable stepper driver as expansion options.

Any help with this decision or thoughts on that matter would be greatly appreciated.

Thank you in advance and best regards

Max