Nikolay-Kha/PyCNC #57: Raspberry PI 4- want to installed PyCNC

djsuthar, 2020-11-17:
Dear Sir,

I  have Raspberry PI4 module and i want to installed PyCNC in the PI 4, but don't know the steps. so please explain the steps.

Thanks

JohnR117, 2021-06-24:
Hi, I think it doesn´t work in Pi4, because you have to make an extension to the Pi4 SoC registers and values for DMA. If I get a Pi4 in the future I would make it

djsuthar, 2021-06-25:
We need to develop CNC controller? Is there anybody support us?

 

Dinesh Suthar

Technical Director

M: 9824303400

 

JDS Motion Technology Pvt. Ltd

H203, Suryam Square, Nr. Karnavati Mega Mall,

Vastral, Ahmedabad - 382418.

 

 <http://www.jdsmotion.com/> www.jdsmotion.com

 

Email :  ***@***.***> ***@***.***,  ***@***.***> ***@***.***