Nikolay-Kha/PyCNC #64: RaspberryPi V3 latest version OS sort of working

MaxWinther, 2021-11-03:
Hi,

Im am trying to run PyCNC  on V3 latest software version. PyCNC seems to work partly, but I also seems loose pulses, and the DMA control Adresse seems to be locked.

Does anybody have experience with PyCNC lately?, any knowledge how to "debug" DMA to verify all pulses has been processed?

regards

Max

Berraud, 2023-02-26:
Hi! I'm having the same issues. I've hooked a logic analyzer and seems like the pulses can't take longer than 15 seconds to be sent or it hangs. We saw some loose pulses in shorter runs too though. Have you been able to find the issue? Long shot, I know. 
Thanks!

Berraud, 2023-02-27:
Damn. Well, thanks for answering! We are going to try an older commit that had worked for us at the time